sccn/clean_rawdata #1: Formula for correlation

Velu44, 2019-09-24:
Dear team,

The line  126 in clean_channels.m script, the formula for computing correlation seems to be missing the mean components (DC offset removal)
	corrs(:,o) = sum(XX.*YY)./(sqrt(sum(XX.^2)).*sqrt(sum(YY.^2)));

I was wondering if it is implemented this way for some valid reasons or if it is a bug to be reported. So, I open this thread. Kindly help me understand this. Thank you.

Cheers,
Velu

arnodelorme, 2019-10-05:
This is because the mean is assumed to have already been removed by the initial high-pass filter (first step in clean_rawdata).

closed 2019-10-05 by arnodelorme

Velu44, 2019-10-05:
Hi Arno,

Thanks for the comment. It's clear now.
However, since I apply the HPF, then segment the data, and therefore I skip the filtering part in clean_rawdata pipeline. I see I will have to modify the correlation formula anyway for my data. :)

arnodelorme, 2019-10-06:
Yes, it is fine to skip filtering if you have already done it. However, the function should be applied BEFORE you segment the data.

Cheers,

Arno

Velu44, 2019-10-07:
Thanks, Arno! 

I have a peculiar situation and I would appreciate any suggestion/feedback from you.

I work with neonates data presented with visual stimuli. There are several portions of data at which the neonates were not quite "attentive" to the presented stimuli. Usually, these portions indicate the newborns moving their heads or crying or falling asleep or simply not looking at the screen. In other words, they typically indicate the very high variance noisy signals and they are not related to the presented stimuli. The timestamps of these portions of data are known to us through a video camera.

Would you still recommend performing ASR before segmenting in this case? 

Sincerely,
Velu

arnodelorme, 2019-10-07:
Dear Velu,

I do not think it will hurt to run ASR as the time range (seconds) is smaller than the one at which you are presenting stimuli. 
But double check which portions are removed to be sure.
Best,

Arno