sccn/clean_rawdata #3: fit_eeg_distribution in asr_calibrate crash

arnodelorme, 2019-12-03:
Transfered from https://sccn.ucsd.edu/bugzilla/show_bug.cgi?id=13115

The line: H = bsxfun(@times,X(1:m,:),nbins./X(m,:)); also crashes when m = 0;

**[reply] [−]DescriptionTyler Grummett 2017-07-18 17:55:02 PDT**

In the clean_rawdata0.32 toolbox, I am having an issue with the fit_eeg_distribution function, which is a function within the asr_calibration function. The error is as follows:

Error using bsxfun
Non-singleton dimensions of the two input arrays must match each
other.

Error in asr_calibrate>fit_eeg_distribution (line 377)
        kl =
        sum(bsxfun(@times,p,bsxfun(@minus,log(p),logq(1:end-1,:)))) +
        log(m);

Error in asr_calibrate (line 180)
    [mu(c),sig(c)] =
    fit_eeg_distribution(rms,min_clean_fraction,max_dropout_fraction);
    
Error in clean_asr (line 164)
    state = asr_calibrate(ref_section.data,ref_section.srate,cutoff);

Error in clean_artifacts (line 219)
    EEG =
    clean_asr(EEG,burst_crit,[],[],[],burst_crit_refmaxbadchns,burst_crit_reftolerances,[]);
    end

Error in clean_rawdata (line 83)
cleanEEG = clean_artifacts(EEG, 'FlatlineCriterion', arg_flatline,...

Looking into it a little further, it appear as though logq only has one row of numbers, so it crashes when it tries to grab logq(1:end-1,:). This occurs when m = 1, n = 6 (X is a 3x6 matrix of numbers), and therefore H is a row of NaNs.

Please note that this is one set of data which didnt work out of a large number of tasks and subjects, so it is working for the vast majority of the time.

Regards,
Tyler

chkothe, 2019-12-04:
The most likely reason for this kind of error is that the data is too short
(at least that was the only condition in which it has ever crashed).

Velu44, 2020-01-14:
Yes, I encountered the same issue when I  used only 250 samples (1 second) as the input. It doesn't throw any error, however, if the samples were increased to 1250 (5 seconds). I confirm Dr.Kothe's comment on this issue.

best,
Velu

arnodelorme, 2022-02-05:
OK, closing because of data too short.

PTDZ, 2022-02-16:
Hi,

Unfortunately, I encountered the same issue and it's a common error (every second or third dataset).  In my case, ASR function works fine when the data are filtered with 1 Hz high-pass filter; and doesn't work (throwing that error described above) when the data are filtered with 0.1 Hz HPF (for some datasets) or 0.01 Hz HPF (for majority, but not all datasets). The sampling rate (256 Hz) and data length is identical (~151000 frames) in all three cases. 

Additionally, it sometimes works for shorter data (~130000 frames, 256 Hz) and thrown that error for longer data. So, I guess it's not only related to data length, but it's a mix of something else (changing the sampling rate to 512 Hz helps, but the issue is probably more complicated). 

Best regards,
PD

arnodelorme, 2022-02-16:
Would you mind sharing some datasets.

PTDZ, 2022-03-11:
Hi,

Yes, I sent two sample datasets to your email address. Same data, filtered with 1 and 0.01 HPF filter. 

Option "Perform Artficat Subspace (...)" works fine with the data filtered with 1 Hz HPF filter, and the described error occurs on 0.01 Hz HPF filtered data (same sampling rate, same length).

Best regards,
PD

PTDZ, 2022-11-08:
@jfrohlich I did not receive any reply in that matter, so I guess the issue was not resolved.

arnodelorme, 2022-11-09:
In the previous report, the data was too short. Would you mind sharing some data?

jfrohlich, 2022-11-10:
In my case, I don't think I have permission to share the data at the moment, sorry ... Perhaps @PTDZ has other data that are sharable and generate the same error?