sccn/clean_rawdata #6: Sample indices to retain in Clean_windows.m function might be wrong

Velu44, 2020-01-14:
File: Clean_windows.m

Line 116: swz = sort(wz);

Since the Z-score values are sorted, the corresponding window indices also change. This change is probably not taken care of, leading to the wrong calculation of removed_samples (line 126). 

[image]

Please confirm. Thanks.
Best,
Velu

arnodelorme, 2020-02-25:
swz dimension seems to be (channel x window) so only channels get sorted not windows. Your comment might be right if there is only one channel. @chkothe would you mind to confirm?

arnodelorme, 2022-02-05:
I have double checked that the code is correct so closing this report.

closed 2022-02-05 by arnodelorme