sccn/clean_rawdata #9: memory usage

rgougelet, 2020-01-29:
clean_asr() splits the data into chunks according to available memory, however the moving_average() function within it ends up operating on the covariance matrix, thereby multiplying the size of each chunk by the number of channels. it also appears that it then doubles the size of the covariance matrix, as well. so, for example, if there are 32 channels, the chunks are 64x too large. this seems to lead to consistent out of memory errors.

simply dividing the chunk size by 2*nchan yields insufficiently small chunks (7 samples, in my case). i attempted to input each channelxchannel vector individually into the moving_average function, but this ends up running too slowly. perhaps there is a simpler solution i'm not seeing.

arnodelorme, 2020-02-25:
@chkothe what do you think about changing the code according to the recommendation above?

arnodelorme, 2022-02-05:
Closing this for now that we have set a fixed amount of RAM (64 Mb). Let us know if this continues to be an issue.

closed 2022-02-05 by arnodelorme