sccn/clean_rawdata #15: Has the availableRAM_GB option been removed or replaced?

cymatn, 2020-07-15:
Hi,

I recently updated to 2.2 from within EEGLAB, and 'availableRAM_GB' now gives the following error.

EG = clean_rawdata(EEG, -1, -1, -1, -1, 20, 0.25, 'availableRAM_GB', 8);
Error using clean_rawdata
Too many input arguments.

It seems to be because ARS_calibrate does not accept it as one possible input anymore. I was wondering if it has been replaced by some alternatives (or default value), or why this had been removed.

Thank you,
Cyrille

arnodelorme, 2020-07-15:
Dear Cyrille,

Which version were you using before? Probably a very old version. There were updates to clean_rawdata that had conflicts, so this option was removed because of that about a year ago. Who told you to use that option? Copying @MakotoMiyakoshi here.

Arno

armenbagdasarov, 2021-01-07:
Hello! I have the same question as Cyrille.

" 'availableRAM_GB' is for clean_rawdata1.10. For any newer version, it will cause error. "

Using the same parameters, the final results from clean_rawdata() fluctuate after ASR. It is demonstrated that adding availableRAM_GB fixes this (in the link below), but now we can't use it if we are using the newer version of clean_rawdata?  

https://sccn.ucsd.edu/wiki/Artifact_Subspace_Reconstruction_(ASR)

Is there any way fix to this? This is a huge problem for replicability. Any help would be much appreciated. 

Thank you! 
Armen

arnodelorme, 2021-01-07:
Could you try resetting the random seed before calling the function. Also better call cleanArtifact directly which has more options.

We will try to add back the option. @MakotoMiyakoshi maybe you could help?

MakotoMiyakoshi, 2021-01-07:
Dear all,

I re-uploaded my final version (ver. 1.10) here as a workaround. The
behavior of the algorithm is identical except for options such as support
for Riemanian geometry.
https://sccn.ucsd.edu/wiki/Artifact_Subspace_Reconstruction_(ASR)#Download_ver_1.10_.28old.29

Arno, unfortunately I am not available for investigating the problem.

Makoto

arnodelorme, 2021-01-08:
I have created version 2.3, which now deliver reproducible results

http://sccn.ucsd.edu/eeglab/plugins/clean_rawdata2.3.zip

Let me know if it all works fine for you and I will release it

armenbagdasarov, 2021-01-08:
Thanks Arnaud! I believe this is working. But it's hard to tell because first I am removing bad channels using clean_rawdata and this is also giving me inconsistent results. I ran the same file 10 times through clean_raw data. Sometimes it says there are 3 bad channels, sometimes 4, sometimes, 5. Is this something similar that can deliver reproducible results for the other functions in clean_rawdata such as removing bad channels? And then I'm assuming this may be a a problem for additional removal of bad data periods function?

arnodelorme, 2022-02-05:
Now closing this report. I have added back availableRAM_GB but not 100% it works as before.

closed 2022-02-05 by arnodelorme