sccn/clean_rawdata #17: Remarks/bugs? clean_windows; asr_calibrate and clean_asr

naloui1, 2020-11-19:
Hello,

I have noticed some bugs in  clean_asr.m :

1)The output of the function clean_windows that is called in clean_asr.m returns all the data and not only the calibration data.

ref_section = clean_windows(signal,ref_maxbadchannels,ref_tolerances, ref_wndlen);  

I think it should be

[ref_section,sample_mask] = clean_windows(signal,ref_maxbadchannels,ref_tolerances,ref_wndlen); 
ref_section.data=ref_section.data(:,sample_mask(1:length(ref_section.data)));  

2)The function asr_calibrate as it is called in clean_asr.m :

state = asr_calibrate(ref_section.data,ref_section.srate,cutoff);

doesn’t contain the input parameter WindowLength ( Window length that is used to check the data for artifact content (Default: 0.5 s))

3) In the function asr_calibrate.m even if we change the value of WindowLength, it will not be taken into account because of (nargin <8) :

 if nargin < 8 || isempty(window_len)
     window_len = 0.5; end

I think it should be 

if nargin < 7 || isempty(window_len)
     window_len = 0.5; end

Can you confirm please.
Thank you
Best

arnodelorme, 2020-12-02:
@chkothe would you mind to comment if that makes sense?

chkothe, 2020-12-02:
Hey, so regarding the following points:

1) the `clean_windows()` function returns the cleaned data -- the `sample_mask` output is purely informational and not required to obtain cleaned data.

2) it would indeed be best to pass the window length into `asr_calibrate` (the parameter is more critical for `asr_process` than it is for calibration, but it is best for them to be matched), and yes, all those nargin checks starting with the one for window_len would have to be shifted by -1. It seems like no one ever tried to override those parameters in the calibration so it wasn't noticed.

naloui1, 2020-12-04:
Hello, thank you for your response.
Regading, the 1st point, the section 'apply selection'  :
'try
    signal = pop_select(signal, 'point', retain_data_intervals);
catch e
.......
end'
doesn't work for me.

I checked the data in the output of clean_windows.m and it was exactly the data that was in the input. So , I added the sample_mask to be sure to get the clean data.
Thank you

arnodelorme, 2022-02-05:
I have fixed the parameter problem

closed 2022-02-05 by arnodelorme