sccn/clean_rawdata #20: I cannot use Clean Rawdata and ASR on a single dataset

BRomans, 2021-02-08:
Hello, I am using eeglab 2021 and I'm trying to use clean rawdata on a single dataset, but the option in Tools is always disabled, regardless all the previous pre-processing steps (removing baseline, filtering, ecc.)
However, if I am working with a study, the option is always enabled. The only problem is that with a study the ASR tool fails because of the N-dimensionality of the EEG object.
I am mostly working from eeglab GUI because I am not very familiar with Matlab, how could I solve this issue? 
I attached 2 screenshots:
- first one shows the disabled option when using a single dataset
- second one shows the error caused by running ASR with a multidimensional EEG object (3 datasets)
Thanks for any help you can provide

[image]
[image]

arnodelorme, 2021-02-08:
This might have to do with menu selection. 

Could you try changing the option "If set, show all menu items from previous EEGLAB versions. You must restart EEGLAB for this to take effect.”

BRomans, 2021-02-08:
I tried both with the option set and unset and it produced no difference. I had the very same problem with eeglab 2020 (I updated today  thinking it was a bug).

BRomans, 2021-02-13:
I think I solved the mistery. My data was epoched, I re-imported the dataset as continuous and now the option is available for use. I was not aware clean_rawdata couldn't work on epoched datasets.

arnodelorme, 2021-02-13:
Good to know. Thank you for the update.

closed 2021-02-13 by arnodelorme