sccn/clean_rawdata #21: "Reject data using Clean Rawdata and ASR" not available

SayaKidby, 2021-03-12:
Hi there, 

I am using EEGLAB v2019.1 on MATLAB 2017b, and wanted to use “Reject data using Clean Rawdata and ASR”.
I’m attaching the screenshot but I can’t seem to select it whichever data I upload, epoched or continuous.
Is this a bug, or is there anything wrong with my data? 
I updated my EEGLAB to v2021.0 but the issue remains. I also tried using it on MATLAB 2019.b but no luck. 

Thanks for your help in advance! 

[image]

arnodelorme, 2021-03-12:
Did you clone EEGLAB from the web? You need to clone EEGLAB with submodules as indicated on the README.md file. Or you can install the cleanrawdata plugin from the EEGLAB plugin managers.

SayaKidby, 2021-03-18:
Hi Arno, 

Thank you very much for your comment, and I am so sorry for the delay in my response. 
I clone EEGLAB from Github with submodules (I think) but the option is not available still. I installed the cleanrawdata plugin and can see it's been added (please see the screenshot) but still when I go to Tools I can't see the Clean Raw data function as available either.  

[image]

Is there anything else I could try? So sorry to trouble you!

arnodelorme, 2021-03-18:
Maybe it is greyed out because you have epoched data not continuous data.

closed 2021-03-18 by arnodelorme