sccn/clean_rawdata #22: Providing resting-state data as the clean reference

CorentinWicht, 2021-04-06:
Dear developers,

This issue is a follow up on an email exchange I got with Arnaud Delorme and Makoto Miyakoshi a year ago.

I wanted to know whether it was possible to provide (for one given subject) `Clean_rawdata` with resting-state data to build the clean reference to finally preprocess ERP data (which are surely dirtier that resting data). Since the idea of the `clean_windows` is to find a _clean reference section_, I figured a resting-state recording would provide cleaner reference sections compared to portions of the ERP data. 

Here is the code I used (**MATLAB R2019b** & **EEGLAB 2021.0** & **Clean_rawdata2.3**):

```
% EEG settings
ERPRAW = XXX % provide here a subject's ERP recording (EEGLAB dataset format)
rsEEG = XXX % provide here a resting-state recording from the same subject (EEGLAB dataset format)
[ALLEEG EEG CURRENTSET ALLCOM] = eeglab; 

% ASR settings
asr_windowlen = max(0.5,1.5*ERPRAW.nbchan/ERPRAW.srate);
BurstCriterion = 10;
asr_stepsize = [];
maxdims = 1;
MaxMemory = 8000; % This is now in MB and not GB!
usegpu = false;
ASR_rsERP = ERPRAW; % Get the EEGLAB structure

% Set seed
rng(0) % For reproducible results 

% Running ASR
% Creating a clean reference section (based on resting data)
% EEGCleanRef = clean_windows(rsEEG,0.075,[-3.5 5.5],1);   
EEGCleanRef = clean_windows(rsEEG); % using defaults

% Calibrate on the reference data
state = asr_calibrate_r(double(EEGCleanRef.data), EEGCleanRef.srate,...
    BurstCriterion, [], [], [], [], [], [], []); 

% Extrapolate last few samples of the signal
sig = [ERPRAW.data bsxfun(@minus,2*ERPRAW.data(:,end),...
    ERPRAW.data(:,(end-1):-1:end-round(asr_windowlen/2*ERPRAW.srate)))];

% Process signal using ASR
[ASR_rsERP.data,state] = asr_process_r(sig,ERPRAW.srate,state,...
    asr_windowlen,asr_windowlen/2,asr_stepsize,maxdims,MaxMemory,usegpu);

% Shift signal content back (to compensate for processing delay)
ASR_rsERP.data(:,1:size(state.carry,2)) = [];
```

The code runs smoothly, unfortunately using resting-state data to select the clean reference section seems to generate more noise that it filters them (**RED = Raw; BLUE = Cleaned**) :
[image]

Does anyone have a hint why this is failing? Maybe this is conceptually wrong?

I currently have this procedure registered in an upcoming Registered Report and would like to dig in deeper to see if I can make it work.

Many thanks,

Corentin

chkothe, 2021-04-07:
If your data are numerically ill-conditioned (e.g., rank-deficient), for instance after re-referencing and/or if the clean data are possibly too short for the given # of channels, then the method could fail to calculate some solutions correctly and that can look like it's injecting noise into the data like in your example. Such a thing has been observed in some rare cases in the past.

CorentinWicht, 2021-04-07:
Dear Dr Kothe,

many thanks for your prompt reply. 

You were absolutely right, using a different set of ERP and RESTING-EEG files solved the issue of sudden bursts of noise. 

I have briefly compared two methods with the new set of files:

1. As presented above, providing ASR with RESTING-EEG data for the clean reference section before running ASR on the ERP recording 
2. The usual method of ASR (i.e. letting ASR detect clean sections directly in the ERP recording)

While I was expecting a more efficient pre-processing while providing RESTING data as clean reference (Method 1), the results actually speak in favour of the usual ASR method (2):
**RED = ERP preprocessed with Method 1**
**BLUE = ERP preprocessed with Method 2**
[image]

On the whole recording, most eye-blinks were left undetected with Method 1 while they were adequately detected and corrected with Method 2. 

Do these results make sense to you? 
I know from experience that ERP recordings can be quite noisy (i.e. eye-blinks, muscle movements, etc. especially outside the component time window) and Resting-States are probably the best way to acquire clean EEG data, hence I am a bit surprised..

arnodelorme, 2022-02-05:
Thank you Corentin, I am now closing this issue.

closed 2022-02-05 by arnodelorme