sccn/clean_rawdata #26: Comments by Makoto

arnodelorme, 2021-06-18:
• The default max_mem is 64 (MB), but in asr_calibrate() it says 'The recommended value is at least 256.' Which is the recommended value?

We should change the recommended value to 64.

	• In asr_calibrate, you use 2^21 to convert MB to Byte. But isn't it 2^20?

I copied the code from the asr_process.m. Looks like it should be 2^20 but Christian put 2^21. I think there is a reason for that (like when you type "df" on Linux you have to divide the memory block by 2 or something). @chkothe any idea?

	• In the main GUI, 'Acceptable [min max] power range' is incorrect. In clean_windows line 105, sqrt is taken.

The function header also mentions power. I am confused @chkothe 

	• This is a reminder in case I forget it tomorrow--when we specify a large value (4096 MB for example) for the max_mem, the final result seems improved. We want to ask why to Christian tomorrow.

Good point.

MakotoMiyakoshi, 2021-06-18:
- Another point is that ASR uses matrix operation heavily. The input data must be full-ranked and double precision. It means typical average referencing and electrode interpolation should not be performed before applying ASR.

arnodelorme, 2022-02-05:
64Mb: changed header
Mb: left as is. There is a reason for it as you can read in asr_process. It is unclear why but better not change it.
Main GUI: changed text
Double precision: the functions already handle that apparently

closed 2022-02-05 by arnodelorme