sccn/clean_rawdata #28: excluding the externals from being impacted by the function

DouweHorsthuis, 2021-11-05:
Hi,
 I was wondering if there is a way to specifically not have externals included in this function. I want to be able to reference back later to these, but I don't see a option to add a channel range. 
As an example I used to use EEG = pop_rejchan(EEG,'elec', [1:160],'threshold',5,'norm','on','measure','kurt');

But I would prefer to use the clean_artifacts function for this.

Thank you!
Douwe

arnodelorme, 2022-02-05:
We have implemented the fix

closed 2022-02-05 by arnodelorme

DouweHorsthuis, 2022-02-07:
Fantastic, thank you so much!

DouweHorsthuis, 2022-04-20:
Good afternoon,
I am currently using EEGLAB V2021.1 and the latest available update (V2.6) of clean_rawdata. And even though it is indeed possible to select a subset of channels. I always get an error after the data is cleaned, or it simply doesn't delete bad channels. 

When I leave all the settings as is, and select all 64channels except for the 8 externals I get the following error:
[image]
(in case it doesn't show it says "Issue with adding back channels removed. Send us your data., This is not a bug (Error occurred in fuction clean_artifacts() at line 298)

To be sure this wasn't something on my side I uninstalled EEGLAB and re-downloaded a fresh one (v2021.1) and updated clean_raw to the last version. In this case:

If I only include the options to delete bad channels, and select 64 chns without externals it gives me the following error

[image]

However, if I run it using all the cleaning functions as they are selected through the gui, it runs without an error. However, even though it says in the command window that channels are being deleted, all 72 channels exist at the end. 

Please let me know if there is anything I can do to stop this issue.
Thank you for your time,
Douwe

MTSVis, 2023-02-14:
> Good afternoon, I am currently using EEGLAB V2021.1 and the latest available update (V2.6) of clean_rawdata. And even though it is indeed possible to select a subset of channels. I always get an [...]

Having the same problem with EGI data.

DouweHorsthuis, 2023-02-15:
I tried it right after the 2022 update and it didn't work for me then, haven't tried it since