sccn/clean_rawdata #33: Investigate effect of RAM on performance

arnodelorme, 2022-02-05:
Makoto report using 4096Mb of RAM yield better results than the default of 64Mb. @MakotoMiyakoshi do you have an example.

MakotoMiyakoshi, 2022-03-03:
I do not have any data for that. Hyeonseok has been actively working on these issues, so he can provide an explanation.

rsomervail, 2023-03-08:
@MakotoMiyakoshi @arnodelorme Is it not necessarily faster to use as much RAM as your machine has available?

MakotoMiyakoshi, 2023-03-15:
Much RAM available means larger chunk of data can be calculated at a time
in a form of larger matrix in the function that has weired line like
8+8+8+8+... to set the default amount of RAM specified.
Hyeonseok has better memory and understanding.

Makoto