sccn/clean_rawdata #34: Unable to see artifact visualization

barlehmann, 2022-02-08:
I keep trying to use the pop_clean_rawdata and have even copied the latest 2.6 version but continue to get errors when trying to use it for viewing the artifacts that have been edited. I thought the issue might be that the dataset I am using has 20 channels but when I pick out both the 20 as well as the 19 subset, neither allows this to work. 

Specifically I get: 
Warning: vis_artifacts failed. Skipping visualization. 
> In pop_clean_rawdata (line 158) 

Any insight/suggestions would be greatly appreciated. TIA. Also, please let me know if this is not appropriate to list this issue here, or in another place, I apologize if this is not the place to do so.

arnodelorme, 2022-03-07:
Sorry for the late answer. We would need to see your data and what you are trying to do. Would you mind sharing your data and instructions on reproducing the issue?

guoyongjian111, 2022-04-17:
[image]
I have the same problem，And I want to know whether clean_rawdata is no longer used after the epoch, such as resting eeg data after the epoch in two seconds。

arnodelorme, 2022-04-19:
Can you share your data?

Yes, clean_rawdata is for continuous data.

guoyongjian111, 2022-04-19:
> Can you share your data?
> 
> Yes, clean_rawdata is for continuous data.

my matlab version is 2021b,eeglabv2022.0, clean_rawdata2.6

guoyongjian111, 2022-04-19:
> 

https://drive.google.com/file/d/1VYrVqncTI7lL2xDpJdT1VPLBr1rwOvUJ/view?usp=sharing
this is a resting eeg data.

arnodelorme, 2022-04-19:
The uncompressed file is not readable by EEGLAB.

guoyongjian111, 2022-04-20:
> 

I know, but uploading is faster after compression. After decompression, it is my data.....

JLULijh, 2023-03-02:
Dude, have you solved your problem? I have the same problem.

arnodelorme, 2023-03-02:
Would you mind resending a file that generates the problem? I could not load the file before because it was corrupted.

JLULijh, 2023-03-03:
Thanks for the reply, I've fixed this issue.

I tried it with the sample_set that came with eeglab.
The problem may be with the eeglab and ASR versions，I ended up successfully visualizing with older versions of eeglab and ASR2.7.

 

李佳辉
***@***.***

arnodelorme, 2023-03-24:
This has been fixed. Let us know if it does not work for you.

closed 2023-03-24 by arnodelorme

Marl1ne, 2023-04-21:
I have the same issue with some but not all of my data files. But by rejecting a small piece of data by eye before running the clean rawdata the visualization works. any idea why its doing so?

arnodelorme, 2023-05-12:
If it does not work with the current version, can you upload a dataset and reopen the issue. This would be very much appreciated.