sccn/clean_rawdata #40: Consult about "vis_artifacts failed" & "Skipping visualization"

samuelsze333, 2022-05-28:
Dear all,

I would like to consult you about a failure to plot rejected data after using Clean_Rawdata and the ASR algorithm. I am going to detail (1) the setting of EEGLAB and MATLAB on my computer and (2) the whole pipeline by the application of Clean_Rawdata and the ASR algorithm.

Setting:
I am using "MATLAB R2022a" and "EEGLAB v2022.0" on a DELL Inspiron 15 5510 that uses Windows.

Procedure:
**1.**  I open MATLAB and type EEGLAB in the command window. (Btw, I cannot figure out how to address the warning. I have checked all folders including the "Toolbox" folder, but I do not see EEGLAB in any folders in the MATLAB folder.)

[image]

**2.**  I import a CNT file / .set file to EEGLAB. (I tried both types of files but still got the warning "vis_artifacts failed. " on the bottom.)

[image]

**3.**  I load channel locations. I choose the MNI coordinate file for BEM dipfit model.

**4.**  I rereference data to the average of the two mastoids, following steps in "https://wiki.cimec.unitn.it/tiki-index.php?page=Data+pre-processing#Re-referencing". That is, I first append the electrode used for online referencing (i.e. CPz) in the "Edit channel info -- pop_chanedit()". Then, in the "pop_reref - average reference or re-reference data" window, I choose "re-reference data to channel(s)", and I choose TP9 / TP10 next to this option. Next, I check "Retain ref. channel(s) in data", and choose "CPz" for "Add old ref. channel back to the data". After that, I go to the same window again, choose "re-reference data to channel(s)", and select both "TP9" and "TP10" for this option. Then, I click "Ok". After rereferencing, the GUI looks like this:

[image]

**5.**  I first low-pass filter the data at 30Hz, and then high-pass filter the data at 1 Hz in the "Filter the data -- pop_eegfiltnew()".

[image]

**6.** In this step, I apply "pop_clean_rawdata()", leaving every parameter unchanged. This means that I do not high-pass filter the data again, but I (1) remove bad channels, perform artifact subspace reconstruction bad burst correction to remove bad data periods, (2) perform additional removal of bad data periods, and (3) enable "Pop up scrolling data window with rejected data highlighted".

After applying "pop_clean_rawdata()", I cannot see the pop-up window showing the rejected data, and I receive this error message: 

> Warning: vis_artifacts failed. Skipping visualization. 
> In pop_clean_rawdata (line 180) 

[image]

Could you please help me with this issue? Thank you so much in advance!!!

Best regards,
Samuel

arnodelorme, 2022-05-28:
Samuel, would you. mind sharing your CNT file?

samuelsze333, 2022-05-28:
@arnodelorme 
Dear Prof. Delorme,

I have just sent you a link to the dataset via email. Please have a look at it. Thank you for your reply and your help!

Best regards,
Samuel

arnodelorme, 2022-05-29:
Thank you. Will look into it.

arnodelorme, 2022-06-01:
This was because you have twice channel Cpz.

closed 2022-06-01 by arnodelorme

samuelsze333, 2022-06-01:
@arnodelorme 

Thank you for your help and pointing out where the problem lies, Prof. Delorme.  I tried to identify the second Cpz, but I can find one only. Please see the channel data plot below. Could you please tell me how I could exclude the second Cpz to prevent the error "vis_artifacts failed"? Thank you so much!

[image]

Best regards,
Samuel

arnodelorme, 2022-06-01:
This is because your script add a second channel CPz 

```
EEG=pop_chanedit(EEG, 'lookup',fullfile(p, 'plugins','dipfit','standard_BEM','elec','standard_1005.elc'),'append',34,'changefield',{35,'labels','CPz'},'lookup',fullfile(p,'plugins','dipfit','standard_BEM','elec','standard_1005.elc'),'setref',{'1:34','CPz'});
```

arnodelorme, 2022-06-01:
I renamed it CPz2 and then was able to get the visualization. I have added a more informative error.

samuelsze333, 2022-06-01:
> This is because your script adds a second channel CPz
> 
> ```
> EEG=pop_chanedit(EEG, 'lookup',fullfile(p, 'plugins','dipfit','standard_BEM','elec','standard_1005.elc'),'append',34,'changefield',{35,'labels','CPz'},'lookup',fullfile(p,'plugins','dipfit','standard_BEM','elec','standard_1005.elc'),'setref',{'1:34','CPz'});
> ```

@arnodelorme 
Dear Prof. Delorme,

I am grateful for your reply.

I am sorry that I forgot to say the data were referenced to CPz online. Because the data were referenced to CPz online, CPz does not show up in the channel data plot. Below is what the channel data look originally after I import the data to EEGLAB. CPz does not appear.
[image]

Therefore, I want to add back the online reference channel CPz. I follow the steps below.

First, I go to Edit > Channel locations, and I choose the default MMI coordinate file.
[image]

Second, in the window, I go to the last channel page (i.e. Trigger), and I click "Append".
[image]

Third, I type "CPz", which is the online reference channel, in the field "Channel label ("label")". Then, I click "Look up locs", and choose the MNI coordinate file again.
[image]

Fourth, I click "Set reference", and type "1:34" in the field "Channel indices" and type "CPz" in the field "Reference".
[image]

I follow the 4 steps above to specify that CPz is the reference channel. These steps result in the line of code you quoted:

> EEG=pop_chanedit(EEG, 'lookup',fullfile(p, 'plugins','dipfit','standard_BEM','elec','standard_1005.elc'),'append',34,'changefield',{35,'labels','CPz'},'lookup',fullfile(p,'plugins','dipfit','standard_BEM','elec','standard_1005.elc'),'setref',{'1:34','CPz'});

Could I ask what I should do to add back the online reference channel CPz in the right way such that the dataset does not have 2 CPz? 

I am so sorry that I have been asking so many questions. I really hope to learn how to solve this kind of problem but I am so inexperienced to solve them myself.

Best regards,
Samuel

arnodelorme, 2022-06-01:
Samuel, Would you mind redirecting your message to the eeglablist since it is not a bug.
Cheers

samuelsze333, 2022-06-01:
@arnodelorme 
No problem. Thank you for taking your time to help me!

arnodelorme, 2022-06-01:
My pleasure

Sacha-Assadourian, 2023-07-31:
Dear Professor Delorme, 

I'm encountering the same problem with the actuals last version of matlab and eeglab, and version 2.8 of clean raw data. I'm loading eeg_data.set from the sample data file and then running clean raw data - ASR. I'm always getting the same error message (please see photos below).

[image]
[image]
[image]

Best regards,
Sacha Assadourian

Sacha-Assadourian, 2023-07-31:
Also, according to a recent video you've upload on youtube 4 month ago, i tried the manipulation on those following version : MatlabR 2021b and eeglab v2022.1. It's still not working.

Best regards,
Sacha Assadourian