sccn/clean_rawdata #44: Problem adding back channels removed

lpc1199, 2022-12-06:
I am trying to do a bad trial rejection using the pop_clean_rawdata function with the following parameters:

      EEG_cortical_BT = pop_clean_rawdata(EEG_cortical, 'channels_ignore', {'HEOG', 'VEOG'},  'FlatlineCriterion','off',...
            'ChannelCriterion','off',...
            'LineNoiseCriterion','off',...
            'Highpass','off',...
            'BurstCriterion',20,...   
            'WindowCriterion',0.3,... 
            'BurstRejection','on',...
            'Distance','Euclidian',...
            'WindowCriterionTolerances',[-Inf 5]);

I want the function to not consider the EOG channels to do this, but for some reason that I can't figure out I get the following error:

  **Error using clean_artifacts (line 320)
  Issue with adding back channels removed. Send us your data.**

Am I doing something wrong or do I have to do some previous step to avoid this error? Here is the Drive link to the signal I am currently using: https://drive.google.com/drive/folders/1LqJiUg7PxSzE_QKXlP8esi7Tl4D5teVC?usp=sharing

arnodelorme, 2022-12-06:
This has been fixed in EEGLAB in https://github.com/sccn/eeglab/commit/1f3dbef35292bb767c3af3060527678f2f44c143
There was some issue with recomputing removed data from Boundary event latency and durations.

closed 2022-12-06 by arnodelorme

pmusfeld, 2022-12-09:
I'm still getting problems with this issue, even when I pull the new fixes you've pointed out in your comments. 
I'm trying to do the same thing, running the clean_rawdata, but by excluding EOG channels. When adding back the removed channel, I'm always running into the following error:

`EEG = pop_clean_rawdata(EEG, 'channels_ignore', {'L_HEOG', 'R_HEOG', 'VEOG'}, 'FlatlineCriterion',5,'ChannelCriterion',0.8,'LineNoiseCriterion',4,'Highpass','off','BurstCriterion',20,...
    'WindowCriterion',0.25,'BurstRejection','on','Distance','Euclidian','WindowCriterionTolerances',[-Inf 7] );`

> Nonfinite endpoints or increment for colon operator in index.
> 
> Error in eegrej (line 94)
>     reject(regions(i,1):regions(i,2)) = 1;
> 
> Error in eeg_eegrej (line 121)
> [EEG.data, EEG.xmax, event2, boundevents] = eegrej( EEG.data, regions, EEG.xmax-EEG.xmin, EEG.event);
> 
> Error in clean_artifacts (line 312)
>         oriEEG = eeg_eegrej(oriEEG, ceil(boundloc));
> 
> Error in pop_clean_rawdata (line 211)
> cleanEEG = clean_artifacts(EEG, options{:});

I also tried to do this manually. I removed the EOGs Channels from the data, filtered the EEG data and used the EEG.etc.clean_sample_mask to filter the EOG data down to the same length for merging. However, the number of points in the filtered EEG set was smaller than the sum(EEG.etc.clean_sample_mask == 1), which also appeared old to me. I assume the number of non-flagged points in the clean_sample_mask should be equal to the number of points in the filtered set, right? 

I am only highpass filtering the data prior to cleaning on 1Hz for ICA. Do you have any ideas what could go wrong? 
Here is my data set: https://drive.google.com/file/d/1UGY17Kcvf977CkD3Wu-wViQGTVXGmL0p/view?usp=sharing

Thanks for looking into this!