sccn/clean_rawdata #47: Possible mistake in clean_asr: window_len parameter does not seem to be passed down from clean_asr to asr_calibrate?

rsomervail, 2023-03-08:
I note that the calibration step of in clean_asr always uses the default 0.5 s window - while the window_len parameter passed by the user to clean_asr is only used for asr_process. 

Is this a mistake? 
I don't fully understand the maths of ASR but shouldn't the thresholds be determined using windows of similar length to those used in the sliding window step of asr_process?

rsomervail, 2023-03-08:
I see this has been addressed in a closed issue from 2 years ago (linked below) - oops, I should have checked first!

I leave this open for now though, because I am dealing with the specific case of ASR in sleep data, where brain signals are typically lower frequency and longer, and the window length seems more crucial for affecting the inappropriate removal of these brain signals.

https://github.com/sccn/clean_rawdata/issues/17#issuecomment-737397502

arnodelorme, 2023-03-15:
Yes, I am closing this report. But feel free to open a new one if you encounter other problems.

closed 2023-03-15 by arnodelorme